rtklibexplorer/RTKLIB #722: Issue using CNES corrections files with RTKPOST

y3nd, 2025-07-20:
Hello

I would like to use RTKPOST along with CNES global correction files to check base coordinates.

https://regina.cnes.fr/en/download-corrections (bottom of the page)

I can retrieve the base data from a .crx file
https://renag.resif.fr/pub/centipede_30s/2025/199/A89200FRA_S_20251990000_01D_30S_MO.crx.gz

From what I understand, .nav should not be needed in PPP with "Precise" Satellite/Ephemeris clock because it will use .clk and .sp3 content.

I always have a "no nav data" message.

[image]

[image]

ourairquality, 2025-07-21:
The broadcast data is currently needed for the satellite clock bias, and there are code paths using the group delays. There is a patch, pephclk related, on the dev branch that overcome this, see https://github.com/rtklibexplorer/RTKLIB/commit/25019279c4bf62e292c623b1db62ce4e161faf . Have not tried this with PPP but this seems to work alone for RTK, and even with predicted precise data (plus some other wip patches) for a quick cold start RTK fix without having to wait for the nav data.

y3nd, 2025-07-23:
Okay
I have tried with rnx2rtkp with the same config file but go a Segmentation fault with the latest commit on dev branch.

ourairquality, 2025-07-23:
Can you try just cherry picking that patch onto the main branch, rather than adopting the entire dev branch. Tried doing this on a local wip branch and the windows rtkpost and rtknavi appeared to ran fine without the nav data, with just the precise sp3 and clk data, fwiw see https://github.com/ourairquality/RTKLIB/tree/oaq