rtklibexplorer/RTKLIB #750: "rtkconv.exe"  (Invalid floating point operation)

LightingSophia, 2025-08-30:
#RTKLIB demo5 V2.5.0 :: There is a bug in "rtkconv.exe" (Invalid floating point operation) . I don't know why.Could you please help me?

ourairquality, 2025-08-31:
Noted that this appears to be from a wip branch and not even from this repo, see the 'EXoaq' version that I added to that wip branch just for such reports! I am not doing releases so guess you built this yourself, and it appears to be the Windows build. If practical can you retry using the 'main' branch. If you can provide a reproducible example then can look at that and likely address the issue. Even better if you can reproduce it using command line tools. If building this yourself then can you run it in the debugger and report the source location that this occurs, and that might be a useful clue, and perhaps a backtrace too?

ourairquality, 2025-09-01:
Thank you. Could not reproduce this using the linux command line convbin app with some local data filtered to just the BDS satellites - still get the start and end times whereas your result has the start and end times uninitialised. Would you be able to provide a link to the input file wuh20670.25o or can you attach this file - perhaps just a few epochs from that file would be enough to reproduce the issue.

```
C   12 C2I L2I C6I L6I C7I L7I C7D L7D C5P L5P C1P L1P      SYS / # / OBS TYPES 
  2025    07    04    01    23   52.0000000     GPS         TIME OF FIRST OBS   
  2025    07    04    01    59   05.0000000     GPS         TIME OF LAST OBS    

```

LightingSophia, 2025-09-01:
I used RAD STUDIO for compilation. The program “rtkcov_b34L.exe” can convert normally, but after compiling “rtkconv_2.5.0.exe,” that error occurs.
IGS data download link: ftp://igs.gnsswhu.cn/pub/gps/data/daily/2025/067/25d/ — ‘ WUH2 ’station.

LightingSophia, 2025-09-13:
I found that a bug occurred at the place shown in the following picture.
[image]

ourairquality, 2025-09-13:
Thank you, shall rework that code to guard against that. This is a RTKLIB extension to encode the standard deviation. Could just replace that it with 'int pstdi = 0' if that extension is not used. It's probably giving a numerical error for log(0), an error not signalled on linux.

ourairquality, 2025-09-13:
Thank you for the specific report. See https://github.com/rtklibexplorer/RTKLIB/pull/755 for the main branch, and the oaq wip branch has also been updated.

closed 2025-09-21 by rtklibexplorer