rtklibexplorer/RTKLIB #751: str2str missing MSM messages when converting input to rtcm3

Stefal, 2025-09-04:
Hi!

Some str2str commands which worked flawlessly with the previous release doesn't work anymore with RTKLib-EX 2.5.0 : The input to rtcm3 conversion, output only 1004 and 1019 MSM messages.

command example:
`./str2str -in ntrip://test:test@crtk.net:2101/LIENSS#rtcm3 -out tcpsvr://:5019#rtcm3 -msg 1004,1005,1006,1008,1012,1019,1020,1033,1042,1045,1046,1077,1087,1097,1107,1127,1230"`

Stefal, 2025-09-06:
Oh Thanks ! It works with `-msg` before `-out`

The str2str help message contains `usage: str2str [-in stream] [-out stream [-out stream...]] [options]`
Shouldn't it be updated to something like  `usage: str2str [-in stream] [[-msg ...] -out stream [-out stream...]] [options]` ?

And maybe add this breaking change in the changelog?

rtklibexplorer, 2025-09-21:
Resolved with update to documentation

closed 2025-09-21 by rtklibexplorer